binsentsu/home-assistant-solaredge-modbus #286: Version 2.0.3 / 2025.2.5 - Finished fetching solaredge-modbus data ... (success:False)

guntherd, 2025-03-03:
From time to time the solaredge modbus integration does completely not work and is not able to fetch data from my inverter. 
This situation can for example have a duration of several hours, before it starts working again.
In such cases the message in the log indicates success:false, from the "Finished fetching solaredge-modbus data".

I have enabled DEBUG. See the excerpt below.
```
2025-03-02 15:17:36.571 DEBUG (SyncWorker_5) [pymodbus.logging] Connection to Modbus server established. Socket ('192.168.1.180', 43246)
2025-03-02 15:17:36.571 INFO (SyncWorker_5) [custom_components.solaredge_modbus] Successfully connected to 192.168.1.161:1502
2025-03-02 15:17:36.571 DEBUG (SyncWorker_5) [pymodbus.logging] Current transaction state - IDLE
2025-03-02 15:17:36.571 DEBUG (SyncWorker_5) [pymodbus.logging] Running transaction 1
2025-03-02 15:17:36.572 DEBUG (SyncWorker_5) [pymodbus.logging] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x87 0x0 0x26
2025-03-02 15:17:36.572 DEBUG (SyncWorker_5) [pymodbus.logging] New Transaction state "SENDING"
2025-03-02 15:17:36.572 DEBUG (SyncWorker_5) [pymodbus.logging] Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2025-03-02 15:17:36.591 DEBUG (MainThread) [homeassistant.loader] Component rpi_power import took 1.284 seconds (loaded_executor=True)
2025-03-02 15:17:36.591 INFO (MainThread) [homeassistant.setup] Setting up rpi_power
2025-03-02 15:17:36.591 INFO (MainThread) [homeassistant.setup] Setup of domain rpi_power took 0.00 seconds
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] RECV: 0x0 0x1 0x0 0x0 0x0 0x4f 0x1 0x3 0x4c 0x5 0x90 0x1 0xda 0x1 0xd8 0x1 0xdc 0xff 0xfe 0xf 0xf5 0xf 0xf3 0xf 0xf9 0x9 0x31 0x9 0x2c 0x9 0x48 0xff 0xff 0xd 0x17 0x0 0x0 0x13 0x8c 0xff 0xfe 0xd 0x20 0x0 0x0 0x9e 0x22 0xff 0xfe 0xd9 0xc 0xff 0xfe 0x0 0xa4 0xf3 0xd6 0x0 0x0 0x10 0x37 0xff 0xfd 0x20 0x3 0xff 0xff 0xd 0x4a 0x0 0x0 0x80 0x0 0xd 0xe0 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x4f 0x1 0x3 0x4c 0x5 0x90 0x1 0xda 0x1 0xd8 0x1 0xdc 0xff 0xfe 0xf 0xf5 0xf 0xf3 0xf 0xf9 0x9 0x31 0x9 0x2c 0x9 0x48 0xff 0xff 0xd 0x17 0x0 0x0 0x13 0x8c 0xff 0xfe 0xd 0x20 0x0 0x0 0x9e 0x22 0xff 0xfe 0xd9 0xc 0xff 0xfe 0x0 0xa4 0xf3 0xd6 0x0 0x0 0x10 0x37 0xff 0xfd 0x20 0x3 0xff 0xff 0xd 0x4a 0x0 0x0 0x80 0x0 0xd 0xe0 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] decode PDU for 3
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] Frame advanced, resetting header!!
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] Adding transaction 1
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] Getting transaction 1
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] [1424, 474, 472, 476, 65534, 4085, 4083, 4089, 2353, 2348, 2376, 65535, 3351, 0, 5004, 65534, 3360, 0, 40482, 65534, 55564, 65534, 164, 62422, 0, 4151, 65533, 8195, 65535, 3402, 0, 32768, 3552, 32768, 32768, 65534, 4, 0]
2025-03-02 15:17:36.623 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\xa4', b'\xf3\xd6']
2025-03-02 15:17:36.624 DEBUG (SyncWorker_5) [pymodbus.logging] Current transaction state - TRANSACTION_COMPLETE
2025-03-02 15:17:36.624 DEBUG (SyncWorker_5) [pymodbus.logging] Running transaction 2
2025-03-02 15:17:36.624 DEBUG (SyncWorker_5) [pymodbus.logging] SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0xfe 0x0 0x67
2025-03-02 15:17:36.624 DEBUG (SyncWorker_5) [pymodbus.logging] New Transaction state "SENDING"
2025-03-02 15:17:36.624 DEBUG (SyncWorker_5) [pymodbus.logging] Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2025-03-02 15:17:36.662 DEBUG (SyncWorker_5) [pymodbus.logging] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2025-03-02 15:17:36.662 DEBUG (SyncWorker_5) [pymodbus.logging] RECV: 0x0 0x2 0x0 0x0 0x0 0xd1 0x1 0x3 0xce 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x0
2025-03-02 15:17:36.662 DEBUG (SyncWorker_5) [pymodbus.logging] Processing: 0x0 0x2 0x0 0x0 0x0 0xd1 0x1 0x3 0xce 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x0
2025-03-02 15:17:36.662 DEBUG (SyncWorker_5) [pymodbus.logging] decode PDU for 3
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] Frame advanced, resetting header!!
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] Adding transaction 2
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] Getting transaction 2
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] [32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 32768, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32768, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32768]
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.663 DEBUG (SyncWorker_5) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2025-03-02 15:17:36.665 DEBUG (MainThread) [custom_components.solaredge_modbus] Finished fetching solaredge-modbus data in 0.140 seconds (success: False)
```

In case this issue is gone, the log file looks like this:
```
2025-03-02 20:52:40.384 DEBUG (MainThread) [homeassistant.setup] Phase platform_setup for utility_meter (140468493415360) took 0.04s (elapsed=0.04s) (wait_time=0.00s)
2025-03-02 20:52:40.384 DEBUG (SyncWorker_0) [pymodbus.logging] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
[... cut ...]
2025-03-02 20:52:40.385 DEBUG (SyncWorker_0) [pymodbus.logging] handle: [b'B\x90', b'q\xc8']
2025-03-02 20:52:40.385 DEBUG (SyncWorker_0) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x04']
2025-03-02 20:52:40.397 DEBUG (MainThread) [custom_components.solaredge_modbus] Finished fetching solaredge-modbus data in 0.299 seconds (success: True)
```

The described problem occurs since several month. I always update my Home Assistant with all updates. 
At the moment HA 2025.2.5 and Solaredge modbus integration with version 2.0.3

maagges, 2025-03-13:
I have exactly the same, currently with Version 2.0.3 and HA 2025.3.2
The debug log says "Successfully connected", but no data available.
`2025-03-13 19:28:58.524 INFO (SyncWorker_8) [custom_components.solaredge_modbus] Modbus client is not connected, trying to reconnect
2025-03-13 19:28:58.526 INFO (SyncWorker_8) [custom_components.solaredge_modbus] Successfully connected to <ip removed>:1502
2025-03-13 19:28:58.580 DEBUG (MainThread) [custom_components.solaredge_modbus] Finished fetching solaredge data in 0.056 seconds (success: False)`

TheRealSlimSchaali, 2025-03-17:
Same for me. I also get the success: False.

picki72, 2025-03-18:
Same here: 

2025-03-18 15:12:22.302 ERROR (SyncWorker_6) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 320, in read_modbus_data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 331, in read_modbus_data_meter1
  File "/config/custom_components/solaredge_modbus/__init__.py", line 480, in read_modbus_data_meter
  File "/config/custom_components/solaredge_modbus/__init__.py", line 160, in validate

Real issue as this impacts EVCC, Energy Dashboard and other stuff