binsentsu/home-assistant-solaredge-modbus #288: Cannot open Solaredge Modbus configuration interface on HA 2025.03

SmolikOne, 2025-03-11:
Hello, after updating HA to Core 2025.3.2, OS 14.2 Solaredge Modbus stopped working. I've removed this integration, installed it again and now I received an error message saying configuration interface cannot be open: "message":"Invalid handler specified". Is there any solution to this issue?

ruffma, 2025-03-12:
Hi, similar for me: I cant see the Modbus integration any longer after updating to 2025.3.2. Neither in settings nor in HACS

AdamWeglarz, 2025-03-13:
the same problem. this isprobable fix! -> Solaredge's Modbus [manifest.json](https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/6a3179cb86063ba92a4457a31ce76ea128d251e7/custom_components/solaredge_modbus/manifest.json#L5) currently contains pymodbus==3.7.4 so you can first try to change it to pymodbus>=3.7.4.

Charly-Man, 2025-03-13:
Changing the manifest.json was not successful.
The error message remains. `{
  "domain": "solaredge_modbus",
  "name": "SolarEdge Modbus",
  "documentation": "https://github.com/binsentsu/home-assistant-solaredge-modbus",
  "requirements": ["pymodbus>=3.7.4"],
  "dependencies": [],
  "codeowners": ["@binsentsu"],
  "config_flow": true,
  "version": "2.0.3"
}`

AdamWeglarz, 2025-03-13:
Did you restart home assistant?

SmolikOne, 2025-03-13:
For me Solaredge integration was visible in HACS, reinstalling it and then modifying the manifest.jason resolved the issue.

AdamWeglarz, 2025-03-13:
Did you verify if there is other integration with this dependency?

Charly-Man, 2025-03-13:
After reinstalling the integration and customising manifest.json, the following error now occurs:  Error updating modbus data: Value 0 failed validation (>0)
I don't have another integration that uses Modbus.

closed 2025-04-08 by binsentsu