binsentsu/home-assistant-solaredge-modbus #292: not able to config

pgas37, 2025-04-05:
I reinstalled HA to have a clean system. I added terminal, file editor, mqtt and HACS. All without problem. 

Then added the home-assistant-solaredge-modbus via HACS. Then rebooted the system. 
Under /homeassistant/custom_components/Solaredge_modbus/ 10 items are available

The installation manual states: install through HACS. After reboot of Home-Assistant, this integration can be configured through the integration setup UI. 
However, under integration & services no option appears to configure the integration. 
In the same menu under devices Solaredge Modbus appears, but that appears to be the "disabled modbus update".
No "unused entities"

Core 2025.4.1
Supervisor 2025.03.4
Operating System 15.1
Frontend 20250404.0

Did i forgot something?
Any idea why I can't configure anything?

Thanks in advance

pawel-tomkiel, 2025-07-02:
Did you try "+ Add Integration"? SolarEdge Modbus should be there for configuration:

[image]

Takkster, 2025-07-02:
@pawel-tomkiel Yes, the above error message appears right after clicking on the integration in your screenshot.

Takkster, 2025-07-07:
I tried again after the update today and it worked 👍