binsentsu/home-assistant-solaredge-modbus #295: Unable to control settings

DavidLeybourne, 2025-04-09:
With the upgrade to .4 the storage control method changes fail, int has no len.

Reverted back all ok

schinzelh, 2025-04-09:
Same here

[image]

drlkp, 2025-04-09:
for me it perform action but after some seconds return to the previous, example not able to disable battery discharge

binsentsu, 2025-04-09:
should be fixed in V2.0.5. Can you confirm?

schinzelh, 2025-04-09:
v2.0.5 works for me, thanks!

DavidLeybourne, 2025-04-09:
Not sure how to upgrade to v2.0.5, but I replaced the init.py with the latest version updated today, and it works again. Plugin still showing v2.0.3 but all seems good.

Thanks for the speedy fix

closed 2025-04-09 by DavidLeybourne