binsentsu/home-assistant-solaredge-modbus #297: HA 2025.4.1 + SolageEdgeModbus 2.0.4/5 causes "Connection unexpectedly closed 0.000 seconds" exceptions

sffbigmac, 2025-04-09:
Starting with `2.0.4` slash `2.0.5` (and at the same time move to **HA 2025.4.1**) I started to see the recently reported exception:
```
Logger: custom_components.solaredge_modbus
Source: helpers/update_coordinator.py:412
integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: 19:43:24 (3 occurrences)
Last logged: 22:10:22

Error fetching solaredge data: Error updating modbus data: Modbus Error: [Connection] ModbusTcpClient solaredgehost:1502: Connection unexpectedly closed 0.000 seconds into read of unbounded read bytes without response from slave before it closed connection
Error fetching solaredge data: Error updating modbus data: Value 0 failed validation (>0)
```

preceeded with a
```
Logger: pymodbus.logging
Source: custom_components/solaredge_modbus/__init__.py:296
integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: 22:09:52 (1 occurrences)
Last logged: 22:09:52

Exception response 131 / 0
```

These seem to happen on startup of HA, but also occur later occassionally. Data at the time of occurrence is missing accordingly from the sensor.

srt4fun, 2025-04-10:
Same error logs here, 2.0.4/2.0.5 broken. HA Core Ver: 2025.4.1

Downgraded to 2.0.3, and all is well, for now.

sffbigmac, 2025-04-10:
Did the same (downgrade only the solaredge modbus integration to 2.0.3 but keep HA 2025.4.1) right now, restarted HA and there is no closed modbus connection failure at startup anymore (will keep monitoring the logs to see if it pops up again over the course of the day) which was realiably there with 2.0.4 and 2.0.5. Strong indication that this does relate to pymodbus 3.8.x compared to 2.7.x.

sffbigmac, 2025-04-10:
PR: https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/299
working for me with HA 2025.4.1
@srt4fun you might also just want to pull this in locally (in case you need the other accumulated fixes for register writing and device id configuration) which is what I did until the version becomes available through hacs

binsentsu, 2025-04-10:
Reverting is not really an option. Home assistant core ships with version 3.8.3 of pymodbus for the default modbus integration. No two versions of the same library can exist within the same process. So if this integration downgrades it can not coexist with default modbus integration activation.

sffbigmac, 2025-04-10:
Ok good to know, I am new to dependency management in HA, so I was under the impression that with the manifest each integration can wire the dependency version as needed for its own purposes. 

Though it would be interesting to know why it is working for me then, also for @srt4fun who effectively achieved the same by reverting to an older version. Is it because still an older version is lingering around, given that we were coming from an older HA version? How and when are these cleaned up?

Besides this we'd of course still need a true fix that works with 3.8.x and doesn't run into the connection timeouts (which it by the way did many times over the course of the night, so it wasn't just an init issue on HA startup).

srt4fun, 2025-04-11:
@sffbigmac No major issues here with 2.0.3. I just use it for logging and a couple simple automations. HA throws a template error for M1, but most M1 readings are reported, so not a huge deal.

dvoijen, 2025-04-11:
Same here, previous version works ok.