binsentsu/home-assistant-solaredge-modbus #302: Active Power Limit entity completely gone

JeroenHaringman, 2025-04-21:
I was running this integration version at the latest version V2.0.5. All sensors I use (power, energy, AC voltage) were working fine.

Sometime last year I had been playing with the Active Power Limit, and it worked fine then. However, when I wanted to work with it today, the Active Power Limit entity was completely gone. Integration reload, HA restart etc. didn't fix it.

Eventually I found this [slightly-related topic](https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/277), and I followed the recommendation there to downgrade to V1.13.0. Active Power Limit is now present again, and works as it should.

So, somewhere between V1.13.0 and V2.0.5 the Active Power Limit functionality broke. 

Please look into this? This will become more important in my country, the Netherlands, because feeding power back into the grid is being discouraged (by financial means) more and more.

JonavdBrink, 2025-05-01:
Sure is! I had it but I could not control the power limit from HA so I'm still digging into it.

ReSpawN, 2025-05-09:
I'm digging into this as well. I came as far as to adding `"power_control":true` to the `data` param in the `core.config_entries`. It's back, however control does not seem to work. I can drag it to whatever I want, it still pops back to 100.

JonavdBrink, 2025-05-10:
So I did some research, you've added or changed writing to F300/F304 in your code? Can you publish full example in that case?

The code you showed from the custom component itself is in const.py right?

Did you change the export_control parts?