binsentsu/home-assistant-solaredge-modbus #319: KeyError: 'modbus_address' since release V2.1.0

Mosibi, 2025-07-06:
I just updated to version v2.1.0 and the integration stops working. It spits out the following

```
2025-07-06 11:51:04.982 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry solaredge_modbus for solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 102, in async_setup_entry
    address = entry.data[CONF_MODBUS_ADDRESS]
              ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'modbus_address'
```

larkoe, 2025-07-06:
Just updated an have the same error

Mosibi, 2025-07-06:
>> My afterthought is that I probably also could just have removed the device from the integration in the GUI and then re-add it.
> 
> Seems that (in my case) removing and re-adding solves the problem... fingers crossed!

Thanks for confirming. This is probably less scary for a lot of people !

larkoe, 2025-07-06:
Can also confirm that removing and re-adding solves the problem for me.

HenriMatthijssen, 2025-07-06:
Above also worked for me.

binsentsu, 2025-07-06:
Reason for this is an oddness in HA where the migrated/updated config is persisted, but the old version is still sent to the setup of the integration. A second reboot should also resolve it.