binsentsu/home-assistant-solaredge-modbus #320: config_entries issue "read_battery3 "

lfeuillet, 2025-07-06:
Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 12:35:17 PM (2 occurrences)
Last logged: 12:37:13 PM

Error setting up entry solaredge for solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 110, in async_setup_entry
    read_battery3 = entry.data[CONF_READ_BATTERY3]
                    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
KeyError: 'read_battery_3'

zorak1103, 2025-07-06:
same problem for me with HA 2025.7.1 and version 2.1 from HACS

janprive, 2025-07-06:
Is this the same issue as in: 
https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/319

klacol, 2025-07-06:
Removing the inverter and re-adding it, solved the issue for me. I just have one battery, no CONF_READ_BATTERY3

binsentsu, 2025-07-06:
Reason for this is an oddness in HA where the migrated/updated config is persisted, but the old version is still sent to the setup of the integration. A second reboot should also resolve it.