binsentsu/home-assistant-solaredge-modbus #321: Problems after upgrade to v 2.1.0

adriederegt, 2025-07-06:
After upgrade to v 2.1.0. I get the error "Failed to setup". Checking the logs I see this:

Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 3:37:37 PM (1 occurrence)
Last logged: 3:37:37 PM

Error setting up entry solaredge_modbus for solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 102, in async_setup_entry
    address = entry.data[CONF_MODBUS_ADDRESS]
              ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'modbus_address'

Modbus sensors are unavailable.
What is going wrong here?

du77an, 2025-07-06:
Same here, I did restart HA...

Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 16:10:02 (1 occurrence)
Last logged: 16:10:02

Error setting up entry solaredge for solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 104, in async_setup_entry
    power_control = entry.data[CONF_POWER_CONTROL]
                    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
KeyError: 'power_control'

pinky12349, 2025-07-06:
Exactly the same issue here.

2025-07-06 16:53:39.236 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry solaredge for solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 104, in async_setup_entry
    power_control = entry.data[CONF_POWER_CONTROL]
                    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
KeyError: 'power_control'

Stash37, 2025-07-06:
Solaredge integration died for me as well after upgrade

AlphaZuluNRW, 2025-07-06:
I have the same problem. One time I wasn't careful and didn't make a backup, and it was fatal.

LAPDragnet, 2025-07-06:
Same here but I luckily did a full backup before and was able to revert the upodate.

AlphaZuluNRW, 2025-07-06:
> > Reboot and add solar edge modbus again. That solved it for me

That didn't work here because I got the message that it was already set up.

AlphaZuluNRW, 2025-07-06:
> Do you have solar edge modbus config in configuration.yaml in that case you my need to remove that first.

Good question! Since the extensive backup is currently being restored, I can't check. But it could be that was the case...

Well, I hope everything will work again after the restore, and then I'll wait and see what happens with the update.

robertvanommen, 2025-07-06:
i deleted the integration and then added it again, it works again now.

binsentsu, 2025-07-06:
Reason for this is an oddness in HA where the migrated/updated config is persisted, but the old version is still sent to the setup of the integration. A second reboot should also resolve it.

pinky12349, 2025-07-06:
Restarting did not help. I removed the divice and added it again. now it wokrs. 
thanks for the fast fix.

satho2210, 2025-07-06:
Following the update, I removed the device from the integration and re-added it using the same predefined name. After restarting, everything functioned as expected.

binsentsu, 2025-07-06:
@AlphaZuluNRW Thanks for pointing out. The new config was indeed not persisted correctly, thereby the reboot did not fix it. Is fixed for future installers in V2.1.2

AlphaZuluNRW, 2025-07-06:
The restore is finally complete, and everything's working fine so far. I'm just not ready to update the integration, especially since version 2.1.0 is still displayed...

adriederegt, 2025-07-06:
I have also re-installed the integration and now it is functioning .

Too bad that some of the sensors now have different names

convicte, 2025-07-07:
After 3 reboots the migration finally took.

Going to 2.1.2 and then reboot until it works, is my recommendation.. 😆

Good luck and thanks for the quick response.

adriederegt, 2025-09-25:
Ls,

I resolved it by re-installing the aad-on

Regards,

Adrie de Regt