binsentsu/home-assistant-solaredge-modbus #326: V2.1.2 Failed setup, will retry: 'SolaredgeModbusCoordinator' object has no attribute 'close'

PascalWenders, 2025-07-07:
All new releases after 2.0.3 I cannot install. It is not possible to connect.

With the version V2.1.2 no install errors but initialisation is not possible.
In Version 2.1.2 I get the error **Failed setup, will retry: 'SolaredgeModbusCoordinator' object has no attribute 'close'**

corelanc0d3r, 2025-07-07:
confirmed, same issue here

binsentsu, 2025-07-07:
That specific close() error mentioned is fixed in V2.1.3.

proza19, 2025-07-07:
Also version V2.1.3. does not work. Where I did a reload of HA

PascalWenders, 2025-07-08:
@noddy21007 do you mean that HA needs two restarts before the V2.1.3 works?

noddy21007, 2025-07-08:
Following installing of the upgrade a restart is required. The integration fails to initialize so as mentioned by others it initializes successfully after a subsequent (2nd) restart.

kalmanatika, 2025-07-09:
It won't work for me! I've restarted it several times. I've even reinstalled version 2.1.0, but it won't work at all! Can you give me some advice?