binsentsu/home-assistant-solaredge-modbus #328: Error migrating entry solaredge for solaredge_modbus

misko903, 2025-07-07:
Hello,

after upgrading and restarted many times (5+), disabling, enabling integration and everything, I sitll got this error:

```
Logger: homeassistant.config_entries
Zdroj: config_entries.py:1098
Prvýkrát sa vyskytol: 17:54:00 (1 výskyt)
Naposledy prihlásený: 17:54:00

Error migrating entry solaredge for solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1098, in async_migrate
    result = await component.async_migrate_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 180, in async_migrate_entry
    if not await hub.read_device_info():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 332, in read_device_info
    data = await self.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        unit=self._address, address=40004, count=64
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 303, in read_holding_registers
    return await self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=address, count=count, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 190, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

Am I the only one now, is there any proper solution, how to repair it?

Thank you!

emilezweep, 2025-07-07:
Same issue.

bjkluft, 2025-07-07:
Updating to the latest also introduced migrating errors in my case:
`2025-07-07 20:16:57.679 DEBUG (MainThread) [custom_components.solaredge_modbus] Migrating configuration from version 1.1
2025-07-07 20:16:57.680 INFO (MainThread) [custom_components.solaredge_modbus] Modbus client is not connected, trying to reconnect
2025-07-07 20:16:57.773 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.xxx.xxx', 1502)
2025-07-07 20:16:57.874 WARNING (MainThread) [custom_components.solaredge_modbus] Not able to connect to 192.168.xxx.xxx:1502
2025-07-07 20:16:57.875 ERROR (MainThread) [custom_components.solaredge_modbus] Failed to connect to hub`

Gavdtas, 2025-07-09:
I have the same issue after the update.

Gavdtas, 2025-07-10:
So I was able to solve it by deleting the integration, adding it again.. And restarting inbetween all the steps. 

First try it did not work, second time it did!

BooBoss, 2025-07-10:
> So I was able to solve it by deleting the integration, adding it again.. And restarting inbetween all the steps.
> 
> First try it did not work, second time it did!

I can confirm it worked for me as well.

1. Delete integration (leave it in HACS - just remove from devices)
2. Restart HA
3. Add integration again

LAPDragnet, 2025-07-10:
> So I was able to solve it by deleting the integration, adding it again.. And restarting inbetween all the steps.
> 
> First try it did not work, second time it did!

> I can confirm it worked for me as well.
>
> 1. Delete integration (leave it in HACS - just remove from devices)
> 2. Restart HA
> 3. Add integration again

So the historical data is still there? No issues with the Energy Dashboard?

BooBoss, 2025-07-10:
> So the historical data is still there? No issues with the Energy Dashboard?

Seems like. Here's graph for 6-7 July of current AC power. From what I know how HA works deleting integration does not mean deleting entities.

[image]

Marsman83, 2025-07-10:
> I can confirm it worked for me as well.
>
>     1. Delete integration (leave it in HACS - just remove from devices)
>
>     2. Restart HA
>
>     3. Add integration again

I can confirm this works, all history is retained so no loss.

LAPDragnet, 2025-07-12:
I got doubts when getting the dialog box for "_Remove the integration_":
I get the warning that says "_All units and entities will be removed permantely_". So something is kept after all?
Waiting for now.

BooBoss, 2025-07-12:
I can only say that for me history remained. It might be true however that entities will be removed but that doesn't mean their history will be removed. If you re-create entities with exactly same names history will be preserved. This is how HA works. If you delete entity history is not removed right away but during database purging task.

Do yourself a test. Create Switch helper, Switch it few times, delete it, recreate it with exact name, check if you see history from before deleting it.

BooBoss, 2025-07-13:
I did it on 2.1.3 (remove, restart, re-add) and it worked for me.

LAPDragnet, 2025-07-13:
> I did it on 2.1.3 (remove, restart, re-add) and it worked for me.

Good, will try again tomorrow 👍

BooBoss, 2025-07-14:
> So, now it is done and it works for me too ☀️😎

Good 👍

danmalone326, 2025-07-15:
> Strange, my inverter is still working on 1502. Haven't you changed it by accident?
I know **I** did not change at the inverter, but I can't explain why it was different.  
Looking in this component code I can see the default has been set to 1502 for at least 6 years.
I did find many documents stating that Modbus over TCP by default runs on port 502.
- https://en.wikipedia.org/wiki/Modbus#Modbus_TCP
- https://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf
Maybe there was a SolarEdge firmware update? Many documents from SolarEdge say they default to port 1502, but this one from June 2025 says port 502.
- https://knowledge-center.solaredge.com/sites/kc/files/sunspec-implementation-technical-note.pdf
It looks like both have been used.

misko903, 2025-07-16:
Solved. Used the restart solution, but issue was:
After opening the port on inverter, the modbus TCP connection must be established within 2 minutes. After first connection, it will remain open. 
There was a bigger FW update on my inverter, that was why I need to do it again probably.

BooBoss, 2025-07-16:
I don't understand one thing. Normally my inverter got some updates via OTA. But when few months ago guy came in for annual inspection (which is expensive AF BTW) he told me that firmware is VERY old and did update via cable. Modbus connection remain working however after this update.