binsentsu/home-assistant-solaredge-modbus #329: Entities disabled "Cannot change status. Disabled by Config entry."

corelanc0d3r, 2025-07-08:
While trying to resolve issues with 2.1.2, I went through a series of disabling/enabling, removing & re-adding my 2 inverters (with the same name).

After updating to 2.1.3 I managed to re-add them into HA. 

However, all entities are disabled.

I am not able to enable them again. I see a message "Cannot change status. Disabled by Config entry." just above the switch to enable them.

bartje01, 2025-07-12:
Having the same issue. Does someone have a fix/solution?

corelanc0d3r, 2025-07-12:
I managed to get it to work again:
(I have 2 inverters)

1. Remove the existing entries 
2. Create new entries, but with new/different names. Confirm they work
3. Remove the entries again 
4. Create again with the original names

bartje01, 2025-07-12:
Thank you for your response! Can you confirm that historical data is not lost?

bartje01, 2025-07-12:
Removing and adding the entry with a new name indeed works. Removing again and creating with the original name results in the same disabled entities again on my end.

bartje01, 2025-07-14:
Does someone else have this issue? I tried removing, adding multiple times without any luck. Still getting the 'entities disabled' message.

Gleichmar, 2025-08-08:
Unfortunately, I have the same problem. When adding solaredge_modbus, all entities appear, but they are all disabled. How can I enable them? Do I have to enter them manually in the YAML? I haven't quite understood that yet. When I install solaredge_modbus_multi, everything works. But I don't want _multi :-)

[image]