binsentsu/home-assistant-solaredge-modbus #332: Versions above 2.0.5 don't work

Daermegil, 2025-07-19:
I'm running 3 instances of home assistant for me and my family. All have solaredge solar panels.
After the upgrade from version 2.0.5 to 2.1.3 (but I tried them all), nothing works anymore.
First you get a migration error. I have read that several restarts would fix this, but that is not the case for me. So I just delete the entity and add it again. But after that the connection immediately fails. The only solution I have so far is deleting the entities, downgrading to 2.0.5 and adding the solar panels again. What am I doing wrong? Do I need specific home assistant version or a firmware update for my converter?

I'm running home assistant in docker (don't know if that is important)
Core 2025.7.2
Frontend 20250702.2

Log:
Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:295
integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: 09:09:23 (3 occurrences)
Last logged: 09:10:24
Not able to connect to [ip adres of converter]:1502

Logger: pymodbus.logging
Source: runner.py:154
First occurred: 09:09:03 (56 occurrences)
Last logged: 09:10:38
Failed to connect [Errno 111] Connect call failed ('[ip adres of converter]', 1502)

[image]

Squixx, 2025-07-21:
I've had the same issue (i think) 

resetting modubs on the _solaredge_ side seems to fix it (enable / disable modbus) atleast for me.

However i'm unsure what causes this, as it seems somehow the integration currently can break modbus

murbanowicz, 2025-08-28:
in my case re-enabling didn't help. I cannot rollback to 2.05 due to issue with pymodbus dependency on new ha :/

AJvdKamp, 2025-08-28:
It again stopped working in the latest HA version. I use HA in a docker container. I can use 2.05 using a **dockerfile** with the following contents:

# Basis image Home Assistant
FROM ghcr.io/home-assistant/home-assistant:stable

# Install pymodbus 3.8.3
RUN pip install --no-cache-dir pymodbus==3.8.3

With **docker compose up -d** the correct pymodbus version is included in the container.

You could also try **SolarEdge Modbus Multi**. That also works for me.

Daermegil, 2025-08-28:
I tried SolarEdge Modbus Multi and that works in my previous and the new version of home assistant. Thanks for the tip @AJvdKamp !
Too bad this integration doesn't work for me anymore. I liked it a bit better than the multi variant.